aceindy/Duepi_EVO #3: Hardware setup issue

fif10, 2021-01-15:
Hello 

Thank you for this great project. 
I am trying to use it  with my stove, but I am having some issues ... on the hardware part. 

I have those one 
- NodeMcu V3 4MB flash ESP8266 12E
- TXB0104 Bi-Directional Level Shifter
with a evacalor stove (PUNTO FUOCO Nelly 9.5 kw + duepi)

So far, I managed to flash the NodeMCU, connect it on wifi, (setup the custom_component on ha.)
On the hardware part I plugged it like this 

[image]

The issue is that if I plug it with the stove, I get the 5V output, but nothing at the output of the level shifter where I should get the 3V.
I also tried to plug the nodeMCU with usb, I get the 3V output, but not the 5V on the level shifter. I guess that it should work as the level shifter is bi-directional
I try also by pluging the OE pin to the low level (3v). still the same...
Did I miss or misunderstood something ? Could you confirm that my setup is ok ? or should I buy and try with another level shifter ... 

Also do I need to configure any thing on esplink for the pin assigment to set the serial pin TX/RX  ?
Once plugged, is there some command I can run directly on esplink console to validate setup ? 
If I send \x1bRD90005f& for instance, should I see the result in the esp console directly ? 

Thanks a lot

aceindy, 2021-01-27:
Hi fif10,

I basically have it like this:
[image]
this is a default setup.
[image]

In ESPLink i use the default config:
[image]

And it the console I can see each transmitted command passing by:
[image]

aceindy, 2021-01-28:
I furthermore believe this has not much to do with Duepi Evo stoves

It sounds more like you have issues with getting ESP-Link working.
Maybe it is better to ask for support [there](https://github.com/jeelabs/esp-link)?

closed 2021-01-28 by aceindy

fif10, 2021-02-01:
Thanks a lot for all the info. That's what I needed.
I will try with another harware devices.