aceindy/Duepi_EVO #4: Information: IR protocoll reverse-engineered

pascaltippelt, 2021-01-27:
I reverse-eingneered the IR-protocoll for the stoves bases on the duepi evo line. Maybe we should cross-reference each other to help searching users. 

[Wamsler-IR-Remote](https://github.com/pascaltippelt/Wamsler-IR-Remote)

aceindy, 2021-01-28:
Awesome work dude..

Not sure if I can use any of it though...as serial uses commands with reply (2 way communication).

I just acquired a serial2wifi module myself as the reverse engineered code from the French guy is incomplete...
Hope I can extract the missing data soon 🥼

closed 2021-01-28 by aceindy

pascaltippelt, 2021-01-28:
The WiFi-Module ist ESP8266-based, at least the one I had in my hands. Maybe you can extract a firmware. If so, I would really be interested in a firmware dump, to analyze it.

aceindy, 2021-01-28:
@pascaltippelt I’ll try to have a look (there is just one thing i lack....time to do it 🤔)

pascaltippelt, 2021-01-28:
A Firmware dump, maybe done by ESP Tools, could help, as we both could work on it! 

But yes, I do understand your time related problem.

aceindy, 2021-01-30:
@pascaltippelt 
Hmm...just opened the unit to have a peek..
It doesn't look like it uses ESP8266 based, but Motorola [RN1723-I/RM](http://ww1.microchip.com/downloads/en/DeviceDoc/70005224A.pdf)
Not sure if I can dump the firmware as easily as ESP....

pascaltippelt, 2021-01-30:
Interesting! So there are at least 2 models around. Could you send me a picture of the pcb?

Does your WiFi module look like this one:
[image]

pascaltippelt, 2021-01-30:
This might help you: https://community.jeedom.com/uploads/short-url/sKZlW8B589uAbgUrlz34R7y4veZ.pdf - it is in french, but there are some information about the interface and some commands.

aceindy, 2021-01-30:
Yea, outside looks the same
[image]

pascaltippelt, 2021-01-30:
Oh wow, definetely a complete different pcb. Mine was way more complex, and contained of cause a different microcontroller. Hope you are successfull!

pascaltippelt, 2021-02-08:
Coult you point out wich pin is 5V, TX, RX, GND? Maybe at the PCB and at the end of the wire (4-pin-connector to stove)? Would be great!

aceindy, 2021-02-10:
1 = White Colour ground

2 = Orange RX

3 = Red TX

4 = Violet +5Volt

pascaltippelt, 2021-02-10:
I expected something like that. 

I am currently working on a ESP-01S based as simple as possible solution to connect to the stove (via serial). At the moment there are only 2 NPN transistors, a line voltage regulator and some resistors and capacitors required. I will update my work as soon as I have anything new to tell.

aceindy, 2021-02-10:
[image]
Got mine from banggood 😉

pascaltippelt, 2021-02-10:
Exactly. But I need some more IO, so I develop myself ;-)

pascaltippelt, 2021-02-12:
I just first time successfully turned the stove off by telnet, using my own adapter board. Works quite well.

I can now confirm this is working on a "WAMSLER WESTMINSTER QUATTRO 6"!

aceindy, 2021-02-13:
i opened a telegram channel, that is probably easier to talk 😉
https://t.me/duepievo