aceindy/Duepi_EVO #5: Persistent values

aceindy, 2021-02-12:
Since the stove does not send any setpoint data, the last used values of setpoint temperature and fan-speed must be saved when shutting down/restarting HA

aceindy, 2021-03-13:
Hmm PascalTippelt seems to have found the command which should retrieve the setpoint, however, for me it always returns 00000020 🤔
Will implement the code anyway 😉

mvroosmalen1970, 2022-03-20:
Found a solution for most commands send: 

<html>
<body>

send | Reponse | decode | Description
-- | -- | -- | --
RC000055 | Value (0) | 0 | File
RC400059 | Hex(20) | ? | ??
RC50005A | 00050025 | 5 | ??
RC60005B | 00160027 | 22°C | retrieve the setpoint
RD000056 | 000F0036 | 15°C | flue gas temperature (133 degs --> 0085002D (ignoring the last 2 digits)
RD100057 | 00B70039 | 18.3°C | ambient temperature
RD200058 | 00060026 | 6 (0-6) | Speed flue gas vent
RD300059 | 00050025 | 5 (0-5) | Speed room vent
RD40005A | 00000020 | 0 (0-5) | Speed pellet
RD70005D | 00000020 | 0 | Temperature S2 water
RD80005E | 01330027 | 1:7 | Cyclus time
RD90005F | 01020023 | load wood | Cyclus phase
RDA00067 | 00000020 | Ok | Errorcode
RDB00068 | 00000020 | 0 | Levelsensor waist
RDC00069 | E6700042 | ? | ??
RDD0006A | 00000020 | 0 | Correction flue gas (TPM*10)
RDE0006B | 00000020 | 0 | Pellet correction
RDF0006C | 00200022 | 32 | Temperature print
REF0006D | 010D0035 | 2690 | TPM
RF00505D | 00000020 | 0 | Temperature S2 water

</body>
</html>

should be useful implementing essential info.

Mark

mvroosmalen1970, 2023-03-14:
I've made some improvements in the code to retrieve the stove setpoint. This fix is working on my stove, this also solves the issue when someone changes the stove setting using the display on the stove.

aceindy, 2023-03-17:
Implemented in test branch, awaiting confirmation from @mvroosmalen1970 before merging

aceindy, 2023-03-17:
> I've made some improvements in the code to retrieve the stove setpoint. This fix is working on my stove, this also solves the issue when someone changes the stove setting using the display on the stove.

In my stove it doesn't seem to work even with the native DP remote app; if I change the setpoint on the stove (or using another app), it does not show on my app

closed 2023-03-17 by aceindy