aceindy/Duepi_EVO #6: Ambient temperature calculation

pascaltippelt, 2021-02-13:
The calculation of the ambient temperature in the pdf is wrong. Only the first 3 chars shall be decoded from hex -> dec an then be divided by 10.

Example for returned result "0C10034&"

01C hex is 193 dec --> 19,3 °C

pascaltippelt, 2021-02-13:
There are more possebilitys for the state of the stove (replies after sending RD90005f&)
01020023 --> load wood
01030024 --> fire on
01040025 --> fire on + vent on

Together with 01010022 ("ignition") this is the whole ignition-process:
1) 0101
2) 0102
3) 0103
4) 0104

Maybe the states are also like the ones from https://github.com/pascaltippelt/Wamsler-IR-Remote ?

aceindy, 2021-02-13:
Looks like there is a system:
RDx000yy
Up til RDA 
 Y = x56 + X 
after that it is
Y = x66 + X

These inquiries I've seen passing by:
RD000056 --> Flugas temperature (133 degs --> 0085002D (ignoring the last 2 digits)
RD100057 --> ambient temperature
RD200058
RD300059
RD40005A
RD70005D
RD80005E
RD90005F --> status
RDA00067 --> error code
RDB00068
RDD0006A
RDE0006B
RDF0006C

Makes me wonder if these are valid:
RD50005B
RD60005C
RDC00069
So far I haven't seen them anywhere....

Most of them can be matched from system overview (just didn't have the time yet to do so):
When loading the page, I can see the inquiries passing by one-by-one ;)
[image]

pascaltippelt, 2021-02-14:
That looks very good. I will test them myself.

The duepi hardware was designed earlier to work with stoves that could also provide hot water. Maybe those unused areas have never been removed since then.

mvroosmalen1970, 2023-03-14:
I've implemented the correct code for the main return states. This is working now. If needed we can implement the other sensors outside the climate sensor.

closed 2023-03-17 by aceindy