aceindy/Duepi_EVO #9: Nolonger availabe in HACS

pascaltippelt, 2021-08-31:
Hi!

I yesterday tried to find yourimplementation in HACS. But I get a "Platform error climate.duepi_evo - Integration 'duepi_evo' not found." error.

I also cant find your implementation in HACS Integrations anymore... Any suggestions?

Best regards

Pascal

aceindy, 2021-09-04:
- [ ] Hey Pascal 👋
I wasn't aware of that, I'll check it out when i have time!
But you can always add it manually by using a custom repository 😉

pascaltippelt, 2021-09-04:
That is of cause what I did, but there are users less able than me, so I thought I should inform you.

aceindy, 2021-09-21:
@pascaltippelt Waiting for reviews 😄 
https://github.com/hacs/default/pull/1069#issuecomment-922429296