aceindy/Duepi_EVO #10: Fan mode doesn't work

mvroosmalen1970, 2022-03-10:
In the current Duepi_EVO the fan option doesn't work. Found the root cause in the replace function x is found twice. The solution is available in my branch/fork. Please review and merge. My pellet stove (duroflame) start and stop correctly, but the fan speed is not changed. After changing it with the plugin to 5 directly after start up it runs as expected.

Marsupoil76, 2024-01-04:
Did you find a solution .?
on my side when I start the heating, and I change the speed, it always comes back to 1

did you have the same problem?

mvroosmalen1970, 2024-01-04:
> Did you find a solution .? on my side when I start the heating, and I change the speed, it always comes back to 1
> 
> did you have the same problem?

Nope it working as expected: 
[image]

Maybe the stove is regulated as its getting close to the setpoint. In the picture above the start is at 3 (after its burns a while) and lower to 1 when the temperature is reached. 

Mark

mvroosmalen1970, 2024-01-05:
> > > Did you find a solution .? on my side when I start the heating, and I change the speed, it always comes back to 1
> > > did you have the same problem?
> > 
> > 
> > Nope it working as expected: ![image](https://private-user-images.githubusercontent.com/36565907/294202747-d1d107de-4ed0-491e-9071-52f04a02b8dd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzODY1NzIsIm5iZiI6MTcwNDM4NjI3MiwicGF0aCI6Ii8zNjU2NTkwNy8yOTQyMDI3NDctZDFkMTA3ZGUtNGVkMC00OTFlLTkwNzEtNTJmMDRhMDJiOGRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA0VDE2Mzc1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1MjQ4MzcwZGE1YjRlZjhkMWEwZjM1ZTA1ZGNhZTVmNzg1ODI3NmRhYTdkMDFiZmY5MWI5Yzk4YTlmMmRhNDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.W1r8RZi9SR953hkqW8Ljx3JOFunB92-x_KG5PElEa_w)
> > Maybe the stove is regulated as its getting close to the setpoint. In the picture above the start is at 3 (after its burns a while) and lower to 1 when the temperature is reached.
> > Mark
> 
> i'm not sure about, it start and it stuck at speed 1/5, even if I start it at speed 2 or 3 or 4 or 5, it always comes back to speed 1

What happens if you do this manually on the stove? You could also look in the the log with debug on as indicated