aceindy/Duepi_EVO #29: Feature idea: No-pellet detection

pascaltippelt, 2023-04-14:
Hi!

I'm not shure if you want to integrate this in your home-assistant integration, but it may be interesting for some users:

Situation: You are running out of pellets and your fire dies. After some 20 minutes the stove cools down an beeps until you refill and reset the error.

Idea: Prevent the faulty state.

Possible solution: Track the exhaust gas temperature. If the stoves exhaust gas temperature is below 95°C for longer than 10 min send a warning message. If the temperature is low for > 15 min stop the stove, before it does it itself.

I testet this with my own system for couple weeks now. It works fine. After 10 min low temp you can still "save" the fire by immediately refilling pellets. After ~20 min below 95°C, the stove sets its state to faulty - so we should act beforehand.

Best

Pascal

aceindy, 2023-04-14:
Hey Pascall

I already have it on auto-reset now...
        # When enabled, auto reset when running out of pellets or ignition failure.
        if self._auto_reset and self._error_code in ("Out of pellets","Ignition failure"):
            await self.remote_reset(self._error_code)

And in HomeAssistant I have an automation checking for any errors and notfy me on my phone :laughing: 

```
alias: Pellet stove issue
description: ""
trigger:
  - platform: state
    entity_id:
      - climate.pellet_stove
    attribute: error_code
    from: All OK
condition: []
action:
  - service: notify.mobile_app_aceindy
    data:
      message: >-
        Check Pellet Stove {{state_attr('climate.pellet_stove', 'error_code')}}
        !!
mode: single

```

pascaltippelt, 2023-04-15:
Well, seems as I was to slow. 😆 Good work!

closed 2023-04-15 by pascaltippelt