aceindy/Duepi_EVO #30: new data

Jimvogelzang, 2023-04-14:
Hi how do i het the new data in my Ha. i have the climate card but also want the fan speed etc. I did the update but i cant find the data should it be in the same climate entity?

aceindy, 2023-04-14:
You use an entity card in Lovelace
```
type: entities
entities:
  - entity: climate.pellet_stove
    type: attribute
    name: Burner Status
    attribute: burner_status
  - entity: climate.pellet_stove
    type: attribute
    name: Error code
    attribute: error_code
  - entity: climate.pellet_stove
    type: attribute
    name: Exhaust fan speed
    attribute: exh_fan_speed
  - entity: climate.pellet_stove
    type: attribute
    name: Flu gas temperature
    attribute: flu_gas_temp
  - entity: climate.pellet_stove
    type: attribute
    name: Pellet speed
    attribute: pellet_speed
  - entity: climate.pellet_stove
    type: attribute
    name: Power level
    attribute: power_level
```

aceindy, 2023-04-15:
@Jimvogelzang Can I close the ticket with this info?

closed 2023-04-15 by aceindy