aceindy/Duepi_EVO #32: Fan mode

Basfoppele, 2023-04-16:
Iam using scripts to call different fan modes in combination with Google Assistant.

This works great! I can tell google this way to set fan speed from 1 to 5

But after the change of 1 to 5 for min to max in climate.py 
I was unable to call HVAC fan mode medium.
Somehow this is to long and not working in a script.

I have changed medium to med in climate.py and now it's working perfect.

Best regards

Bas

aceindy, 2023-04-16:
Send a pull request in github?

Basfoppele, 2023-04-16:
> Are these scripts on HomeAssistant?
> 
> If possible, send a pull request in github?

Yes these scripts are on home assistant.
I will send a pull request later and will try to show what I mean 😉

aceindy, 2023-04-16:
I understand what you want to change, I am just curious about the scripts though, as I fail to understand why the string "Medium"would be too long for a script in HA...or is GoogleAssistant the culprit?☹️ 

(and I am asking for a PullRequest just to do it the 'official way' 🧑‍🏭

Basfoppele, 2023-04-16:
To be honest I don't know how the pull request works because I never used GitHub much before.

I can't give an answer to why medium not seems to work and med does.
But it is in HA,  because if I run the script manual in Ha it is also not working.

This is a copy of my script

alias: Pelletkachel op 3
sequence:
  - service: climate.set_fan_mode
    data:
      fan_mode: Med
    target:
      entity_id: climate.pelletkachel
mode: single

aceindy, 2023-04-16:
Hmm..I geberuik het zelfde in Node-Red
[image]
En geen probleem met fan_mode medium :frowning: 
[image]

Maar ik zal het straks ook eens proberen met een script

aceindy, 2023-04-16:
Hmm..
Ik heb er geen probleem mee...

```
alias: Pellet_medium
sequence:
  - service: climate.set_fan_mode
    data:
      fan_mode: Medium
    target:
      entity_id: climate.pellet_stove
mode: single
icon: mdi:fireplace

```

```
Executed: April 16, 2023 at 19:23:18
Result:
params:
  domain: climate
  service: set_fan_mode
  service_data:
    fan_mode: Medium
    entity_id:
      - climate.pellet_stove
  target:
    entity_id:
      - climate.pellet_stove
running_script: false
limit: 10

```

Home Assistant 2023.4.4
Supervisor 2023.04.0
Operating System 9.5

aceindy, 2023-04-16:
It sounded weird already; i also have 2 ac's, and they also use 'Medium' in their fan_mode 🤔
[image]

Basfoppele, 2023-04-16:
Okay very strange that it didn't work for me? 
Maybe it is something in my Home Assistant?
Anyway i changed it for myself and if it doesn't cause any problems for other it's okay
 Thanks for your time and help 😊

aceindy, 2023-04-16:
I've ask some more users on Telegram to test...
Prettige avaind !

closed 2023-04-16 by aceindy