aceindy/Duepi_EVO #37: Centrometal serial protocol?

kentaur9, 2023-10-07:
Have you been able to integrate to centrometal serial interface? Do you have some details on the protocol it uses?

aceindy, 2023-11-24:
Centrometal? Is that a brand or what ?
My stove uses a Duepi Evo module, which can communicate with an optional wifi module.
I have replaced the wifi module with an esp-link module (lan2ser)

pascaltippelt, 2023-11-24:
The Centrometal Z6 looks like my Wamsler Pellet-Stove. They _should_ use the same Duepi Evo board. Best way to find out: Open the right cover AFTER unplugging the power supply. If something like this is inside, you should be lucky https://larel.com/b2b/en/control-system-board/801-14710080.html:

aceindy, 2023-11-24:
And the initial protocol is described here https://github.com/aceindy/Duepi_EVO/blob/main/InterstoveRemoteCmd.pdf

In the meantime we found a lot more...

closed 2023-11-29 by aceindy