aceindy/Duepi_EVO #39: Can't get it to work, component won't connect

Sholofly, 2023-12-31:
Hi, i'm trying to.connect my artel julia next 2 but i cant get it to work. The component says
```
Time out when polling host
```

My set-up:

Artel Julia Next 2
Wemos d1 mini with esp easy.
[image]
[image]
[image]

On wemos Orange goes to tx red goes to rx...

Anybody an idea what i'm doing wrong?

mvroosmalen1970, 2024-01-05:
port should match yaml.

you have 23 in the yaml and 1234 in wemos

Sholofly, 2024-01-05:
I think ive tried to make them the same. But I'll try again. TX should point to RX and vice versa, isn't it?Just to be sure ;)

mvroosmalen1970, 2024-01-05:
Yes. Follow the schematics

Sholofly, 2024-01-06:
Where can I find the schematics?

mvroosmalen1970, 2024-01-08:
Duepi_EVO/board.jpeg at main · aceindy/Duepi_EVO · GitHub (https://github.com/aceindy/Duepi_EVO/blob/main/board.jpeg)

aceindy, 2024-01-09:
> should point to RX and vice versa, isn't it?Just to be sure ;)
Depends a bit from vendor to vendor, if not f=working, just try to swap....

I will close this, as it is not an ' issue'  
Pleas continue on [telegram](https://t.me/duepistove) or [HA forum](https://community.home-assistant.io/t/pellet-stove-duepi-evo/255841?u=aceindy)

closed 2024-01-09 by aceindy

Sholofly, 2024-01-09:
Yeah, allright with me. Still cant get it to work even with correct port. Tried both options. Thanks anyway.