aceindy/Duepi_EVO #40: Beginners issue

ptitchau, 2024-01-12:
Hi Aceindy. First, many thanks for your great job here!
I am struggling with the use of the integration and I have a very basic issue, not sure if here is the better place to ask sorry... I have all set all the yaml configuration, custom components, etc as you suggested, HW is wired as expected, but I am unable to even see the climate entities in my HA ... :(

Would you have a suggestion please? It will be highly appreciated, I am trying since weeks to control my pellet heater for the winter :)
thanks in beforehand!!!!

Marsupoil76, 2024-01-17:
je suis désolé mais vous nous donnéez que trop peut d'information, quel est le systeme chargé sur votre ESP.? qu'avez vous mis dans votre yaml ? de quel yaml parlez vous .? configuration.yaml..? ou le yaml de l'esp .?

ptitchau, 2024-02-04:
I have trying with esphome and esplink. I'm talking about configuration.yaml. Is it necessary to have the connection and communication with the stove(by the serial port) to see the climate in the entities, or just a good configuration.yaml and good connection to the esp, allow to see the climate ?

aceindy, 2024-02-04:
Good question... I think there should be communication before it will show as entity...but to be honest, never paid attention to it.

I guess the best way to check is to set the log level to debug:
`
logger:
  default: error
  logs:
    custom_components.duepi_evo: debug
`
It should spit out something like this:
`
2023-11-21 11:51:11.280 DEBUG (MainThread) [custom_components.duepi_evo.climate] Pellet stove: Received burner: Eco Idle, Ambient temp: 23.6, Fan speed: 1, Flu gas temp: 58, Exh fan speed: 2820, PelletSpeed: 0, Error code: All OK, Setpoint temp: None
2023-11-21 11:51:11.286 DEBUG (MainThread) [custom_components.duepi_evo.climate] Pellet stove Setpoint retrieval not supported by this stove, using temp_nofeedback 16
2023-11-21 11:51:42.613 DEBUG (MainThread) [custom_components.duepi_evo.climate] Pellet stove: Received burner: Eco Idle, Ambient temp: 23.5, Fan speed: 1, Flu gas temp: 56, Exh fan speed: 2820, PelletSpeed: 0, Error code: All OK, Setpoint temp: None
2023-11-21 11:53:10.457 DEBUG (MainThread) [custom_components.duepi_evo.climate] Pellet stove: Set hvac mode to off
2023-11-21 11:53:12.017 DEBUG (MainThread) [custom_components.duepi_evo.climate] Pellet stove: Received burner: Cooling down, Ambient temp: 23.4, Fan speed: 1, Flu gas temp: 53, Exh fan speed: 2820, PelletSpeed: 0, Error code: All OK, Setpoint temp: None
`
More info here --> 
https://community.home-assistant.io/t/pellet-stove-duepi-evo/255841/110?u=aceindy

I will also close this issue, as it is not a code issue related to GitHub.

aceindy, 2024-02-04:
Please continue on https://community.home-assistant.io/t/pellet-stove-duepi-evo/255841?u=aceindy is more advise is required

closed 2024-02-04 by aceindy

ptitchau, 2024-02-04:
Yes, thank you for your work. I measured 24volts on a serial pin, I have problem on my serial port. You right to close my issue.
Thank you.