aceindy/Duepi_EVO #49: Is it equivalent of using MicroNova based pellet stove with esphome?

freakontrol, 2024-12-06:
Hi, thank you for your work and time! 
I was wondering if the protocol of this project is the same as the one used for the micronova platform in esphome, or are for different pellet stove central units?

aceindy, 2024-12-07:
Hi FreakControl,

Not familiar with the protcol used by MicroNova, but I suspect it will be completely different from Duepi.
I know on https://stufapellet.forumcommunity.net/ there are some controllers for MicroNova, maybe worth to check that out?
(I used google translate to view the forum)

freakontrol, 2024-12-07:
Thank you for the info! I will look in to it!

aceindy, 2024-12-08:
https://stufapellet-forumcommunity-net.translate.goog/?t=57458738&_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp

closed 2024-12-08 by aceindy