aceindy/Duepi_EVO #50: Unable to set temperature below 16°C using Home Assistant integration

jbjardine, 2025-01-01:
Hello, first of all, thank you so much for your fantastic work on this integration that allows controlling the pellet stove via ESP and Home Assistant. It works great overall! However, I have an issue with setting the temperature below 16°C. When I use the official MyDPRemote app, I can successfully set the stove’s setpoint to as low as 10°C. But through this Home Assistant integration, any attempt to go below 16°C is ignored by the stove, and the temperature remains at 16°C minimum.
Everything else is the same hardware and setup, so I’m not sure why the stove will accept 10°C in MyDPRemote but enforces 16°C with this integration.
Any help or insight would be greatly appreciated!

aceindy, 2025-01-02:
Hi @jbjardine ,

I can reproduce the issue (sorry for not noticing it myself, as I use 16 degs as minimum.

I will try to have a look at it, but, as I have to abandon my stove soon (as I'm moving to a new house without my stove), I do not have a lot of time.

jbjardine, 2025-01-02:
Hi @aceindy ,

Thank you so much for your quick reply! I’d be happy to help debug this issue if you have a moment. One idea is to intercept or monitor the commands that the manufacturer’s app sends when setting the temperature below 16°C. If we can somehow compare the data payload from MyDPRemote and from our integration, we might pinpoint the exact difference.

For instance, we could try using a network proxy or some form of logging to capture requests and responses. From there, we could check whether the stove receives a slightly different command or parameter structure when the temperature is set under 16°C via the manufacturer’s app. Let me know if you think this approach might help, and I’ll do my best to assist!

aceindy, 2025-01-02:
@jbjardine That is easy; I wrote a [proxy.py](https://github.com/aceindy/Duepi_EVO/blob/main/evo-python/Proxy.py) to capture the traffic...
my issue is finding time 🤔

jbjardine, 2025-01-02:
Hi @aceindy,

Thank you so much for your incredible insights and support! Following your advice, I was able to debug and pinpoint the root cause of the issue. It turned out to be a double problem. First, there was a hexadecimal encoding issue when setting the temperature below 16°C. Additionally, I discovered an offset discrepancy: it’s 97 when the temperature is below 16°C and 75, as you originally implemented in your code, when it’s 16°C or higher.

I noticed that you created a new branch 8 hours ago that already addresses the hexadecimal encoding issue—great work on that! However, it doesn’t yet account for the offset change depending on the temperature. I’ve submitted a couple of pull requests (#51  and #52) on your older branch with fixes that handle both issues. I’ve thoroughly tested them, and they are working perfectly! Please feel free to review and validate them when you have time.

Once again, thank you so much for your brilliant proxy code—it has been a lifesaver and will be incredibly useful in my day-to-day workflow. I’m genuinely happy to have been able to contribute a little to this project and resolve this issue with your help.

Looking forward to your feedback!

jbjardine, 2025-01-02:
Hi @aceindy,

Apologies for the confusion—I accidentally closed the ticket before your validation! 🙈 I’ll leave it to you to have the honor of closing the ticket once you’ve had the chance to review and validate my modifications.

Thanks again for your incredible support and this great collaboration—it’s been a real pleasure working on this! 😊

aceindy, 2025-01-03:
G'morning @jbjardine , 
Looks like we came up with the similar conclusion 
if (set_point_int < 16):
    	code_hex_str = hex(set_point_int + 97)
Too bad I didn't notice you PR until now...look like your code is 'cleaner' then mine...testing now ;)

aceindy, 2025-01-03:
@jbjardine V2025.01.03 released...thanks...

closed 2025-01-03 by aceindy