aceindy/Duepi_EVO #53: Co-ownership of Duepi_EVO repository

aceindy, 2025-01-03:
Hello @jbjardine @mvroosmalen,

As I am moving to a new 'energy-neutral' house in the near future, I, unfortunately, will have to abandon my pelletstove in  the near future (:sob: ).
Maybe in the future I will obtain another Stove, but that depends on if the 'energy-neutral' house will be really energy neutral :smile: 

But without a pelletstove it will be hard to maintain this repository, so I was wondering if either or both of you are interested in co-ownership (as both of you contributed in PR's)?

mvroosmalen, 2025-01-03:
Hi,
 
No problem,but depending on my adventure with ESPHome (not as uart terminal but as fully functional integration), I'll move to ESPHome as I think it could be a better solution ;-).  
 
Mark

>
> Hello  @jbjardine (https://github.com/jbjardine)   @mvroosmalen (https://github.com/mvroosmalen),
>
>
> As I am moving to a new 'energy-neutral' house in the near future, I, unfortunately, will have to abandon my pelletstove in the near future (:sob: ).
Maybe in the future I will obtain another Stove, but that depends on if the 'energy-neutral' house will be really energy neutral 😄

aceindy, 2025-01-03:
@mvroosmalen

A, yes, this has been on my 'wishlist' for a while...but I won't have any time to make it so....
Looking forward to see it though :wink:

jbjardine, 2025-01-03:
Hi @aceindy & @mvroosmalen,

It would be a pleasure to co-own the repository! With the other contributor, we could split the focus between ESP Link and ESPHome integration. What do you think about this approach?

jbjardine, 2025-01-03:
Thank you for the explanation! Do you think we could implement these changes directly within this repository, or would it make more sense to create a new one specifically for the native ESPHome integration?

aceindy, 2025-01-03:
@jbjardine 
It would be totally unrelated to this HACS integration, so it makes sense to create a separate repository for this.
Maybe even fork the [ESPHome repository](https://github.com/esphome/esphome/tree/dev/esphome/components), and make PR's there 🤔

jbjardine, 2025-01-03:
That sounds like a good approach. However, I’m not very familiar with ESPHome yet, so we’d need to check how we can adapt our code to make it functional and compatible with their ecosystem. Also, we’ll need to figure out how to integrate it into their repository if that’s the route we choose.

aceindy, 2025-01-03:
I use plenty esphome devices, but writing code for it is something else 🤔

closed 2025-01-07 by aceindy

aceindy, 2025-01-17:
> Thank you for the explanation! Do you think we could implement these changes directly within this repository, or would it make more sense to create a new one specifically for the native ESPHome integration?

@jbjardine it seems @mvroosmalen1970 managed to get it working on ESPHome...
I'll try it asap to see how it works :D

mvroosmalen1970, 2025-01-17:
Hi

I created a new git for the ESP home. Is clearer for everyone I think

Let me know if you tested it

> @jbjardine (https://github.com/jbjardine)  it seems  @mvroosmalen1970 (https://github.com/mvroosmalen1970)  managed to get it working on ESPHome...
I'll try it asap to see how it works :D
"Organization", "name": "GitHub", "url": "https://github.com" } } ]

jbjardine, 2025-01-17:
@aceindy @mvroosmalen1970 

Congrats on launching the ESPHome branch—super cool to see it in action. Just a heads-up: I’ve fixed another offset issue for setpoints above 25 °C, so be sure to pull the latest code. Keep me posted on how it goes!