aceindy/Duepi_EVO #54: Unable to set temperature after 25°C using Home Assistant integration

lolo130578, 2025-01-06:
Hello,
it's the opposite of the problem:

https://github.com/aceindy/Duepi_EVO/issues/50#issue-2765297340

It's impossilble to set temperature after 25°c with home assistant
it's good with MyDPRemote
thanks

aceindy, 2025-01-07:
@jbjardine Do you still have the system in test setup?
Looks like we have a similar issues with temp >25 🤔

jbjardine, 2025-01-08:
@aceindy I will take a look this week-end 👍

jbjardine, 2025-01-11:
The spark plug on my pellet stove broke today... I'm already changing it next week and then I'll be able to solve this bug.

aceindy, 2025-01-11:
Ah, that happened to me too...twice already :astonished: 
Manged to find my replacements on Amazon for cheap :smile:

jbjardine, 2025-01-17:
Hey @aceindy 

Honestly, I still don’t get why anyone needs more than 25 °C (especially with energy bills these days!), but I’ve finally repaired my pellet stove, so I could dig into the issue and confirm the cause. Turns out the manufacturer uses different offsets depending on the setpoint range—so above 25 °C, the offset changes and our code was sending the wrong value.

Here’s the quick summary :
	•	For 16…25 °C, offset is 0x4B (75).
	•	But from 26 °C on, it can jump to 0x52 (82) or even 0x3C (60) for higher temps.
	•	That means if we keep blindly using 0x4B (75) for anything above 25 °C, the stove just ignores us.

I’m opening a new branch to handle these multiple offsets correctly (i.e., mapping each temperature range to its matching offset). Once that’s in place, you’ll be able to heat up your living room to 30 °C…if that’s your thing! 😉

(Will link the fix branch once it’s up.)

jbjardine, 2025-01-17:
#55 #56 

All done!

closed 2025-01-18 by aceindy