aceindy/Duepi_EVO #57: Calculating the last two caracters

NenesFab, 2025-02-09:
FOR YOUR INFORMATION (as your program works) :

Quite sure it isn't the right place to contact the contributors, but as I'm a newbie in github, haven't found another way. I also try to control my pellet stove, so I look everywhere to find some intel's. As a very old but not pro programmer, I was very intrigued by the two last digit of the command and response protocol.  

After some testing (calculations), I've found that the commands and reponses of the protocol is only 6 caracters, the last two is a checksum. E.g. response 00000020 is response 000000 + last two digit of the ASCII sum of (30h + 30h + 30h + 30h + 30h + 30h = 120h) => so the last 20 is the BCD for 20h, for the commands it's the same thing so to set the stove at 33°C, the command is "RF2210" with the checksum (52h + 46h + 32h + 32h + 31h + 30h = 15Dh) so the last two caracters are 5D in BCD, and the complete command is RF22105D.

This is why you have to recalculate your "offset" 15°C is 0Fh (ASCII sum 30h + 46h = 76h) but 16°C is 10h (ASCII sum 31h + 30h = 61h), 76h - 61h = 21 in decimal + 1 for the "real" offset in degree, so it's 22 in décimal, so you must change your offset by substracting 22.

Same for  25°C (19h, ASCII sum 31h + 39h = 6Ah) and 26°C is (1Ah, ASCII sum 31h + 41h = 72h), 6Ah - 72h =  - 8 in decimal + 1 for the "real" offset in degree, so it's -7 in décimal, so you must change your offset by adding 7. And so on...

NenesFab, 2025-02-09:
Sorry, little mistake it isn't the BCD but the Hex to String....

NenesFab, 2025-02-09:
Sorry, it isn't BCD, but I don't remember the name of the process to take the first and last four bits of a byte and convert then in ascii numbers and letters..... E.g. 2Dh becomes the ascii string 32h + 44h or 2D in ASCII caracters...

aceindy, 2025-02-09:
Hmm..intresting....

I think it is called "nibble encoding"  😆 

Not sure if I have the time to figure it our, but maybe @jbjardine or @mvroosmalen1970  can do something with it 😉

aceindy, 2025-02-10:
@NenesFab Looks like you nailed it dude...

`def calculate_checksum(command):
    total_sum = sum(ord(char) for char in command)
    lsb = total_sum & 0xFF 
    return format(lsb, '02X')`

I'm implementing this...might take a little while :P

jbjardine, 2025-02-10:
Wow, this is amazing! First of all, huge thanks to @NenesFab for uncovering the whole “nibble-encoded ASCII sum” behind those last two characters. It’s mind-blowing to see how all these offsets are actually part of a proper checksum mechanism!

Also, a big shout-out to @aceindy for the initial code snippet handling the ASCII sum. This could be a real game-changer for controlling every aspect of the stove. Since this logic seems to apply to all commands (not just temperature), we could make a big step forward by implementing this across the board.

@aceindy, how would you like to proceed? Do you prefer that we open a PR with these changes, or do you want to integrate it yourself first? This might require some rewriting throughout the code, so let us know what you think...

Below is a minimal approach that can be adapted for any command. It generates the 6-character “base” command, computes the checksum by adding all ASCII values, and appends the 2-digit hex result:

```
def duepi_checksum(base_cmd: str) -> str:
    total_sum = sum(ord(c) for c in base_cmd)
    lsb = total_sum & 0xFF
    return f"{lsb:02X}"

def build_command(base_cmd: str) -> str:
    csum = duepi_checksum(base_cmd)
    return f"\x1b{base_cmd}{csum}&"

def build_temperature_command(temp: int) -> str:
    temp_hex = f"{temp:02X}"
    six_chars = f"RF2{temp_hex}0"
    return build_command(six_chars)
```

What do you think?

aceindy, 2025-02-10:
@jbjardine I haven't got much further yet then confirming the checksum is calculated correctly.
And yes, it is the same for all commands.

As I will be moving out of my house (and leave the pellet stove😢), i have little time (and even less time left), so go for it if you have the time 👍🏽

jbjardine, 2025-02-10:
@aceindy! No worries at all, I’ll test this function right away and see how it goes. Once everything is confirmed to be working, I can handle the integration and open a PR if you like. Good luck with the move—sorry you have to leave your pellet stove behind!

aceindy, 2025-02-10:
@jbjardine I hope in due time I'll be able to get a new one 😆 
(but that might take a year...or even two 🤔 )

NenesFab, 2025-02-11:
Hi @aceindy @jbjardine @mvroosmalen1970, I'm glad I was able to help, but be careful, now that you have the checksum to make any valid command (and also create/test new ones), in my search I've also seen that there is a duepi firmware programmer (google "loader usb duepi") that use the same serial interface as the WIFI remote.  So there's a high risk of "bricking" your stove whith undocumented commands !

aceindy, 2025-02-11:
@NenesFab Oh,I know about that software. I have been thinking wanting/searching for it for a long time, but never found it 🤨

aceindy, 2025-02-11:
@NenesFab Also happen to have an activationcode ?

NenesFab, 2025-02-11:
@aceindy @jbjardine
Nope, no activation code !
But maybe there are some unknown commands in the dpcom program 😉

aceindy, 2025-02-12:
@NenesFab , @jbjardine, @mvroosmalen1970

Something like this?

aceindy, 2025-02-12:
Hmm...interesting...
The PowerON/Off and Level are actually one and the same command.. 😃 

Power =0 --> Off
Power > 1 --> On, Min Power
Power = 5 --> On, Max Power

SET_POWEROFF =     "F0000"
SET_POWERON =      "F0010"
SET_POWERLEVEL =   "F00x0"

aceindy, 2025-02-12:
Also here something fishy is going on:
SET_AUGERCOR =     "D5000"
SET_EXTRACTORCOR = "D5000"   
SET_PELLETCOR =    "D5000"
(These parameters are currently not used)

NenesFab, 2025-02-13:
@aceindy
> Hmm...interesting... The PowerON/Off and Level are actually one and the same command.. 😃
> 
> Power =0 --> Off Power > 1 --> On, Min Power Power = 5 --> On, Max Power
> 
> SET_POWEROFF = "F0000" SET_POWERON = "F0010" SET_POWERLEVEL = "F00x0"

Yep and for some stove 6 is AUTO power mode

aceindy, 2025-02-13:
Might as well keep this open for discussion :P

NenesFab, 2025-02-16:
It's a pity that we don't have an activation code for the program. In the manual I saw that it was possible to program, edit, save the technical parameters of the stove (UT04 -> A9 -> PR00 etc). Sniffing the serial com between the stove and the program and see how it's done would be nice... When I was young decompiling a program was "quite" easy, just a few thousand assembler codes. Today it's millions... So I don't know, but to start I would use a hexadecimal editor (like hexplorer) and I would look for the text 00000020 in the program, it could be interesting to see the result !?

jbjardine, 2025-02-24:
@NenesFab 
Unfortunately, the pellet stove parameters in the software aren’t accessible because a DLL file is missing from the installation—and I can’t locate "Schede.DLL" anywhere…

NenesFab, 2025-03-19:
@aceindy @jbjardine For your information -- USE AT OWN RISK ---  (Tested on controller BOARD V9 AMG 410615 V9.2.7 made by Duepi, Duepi LED may also be the same !?)

# RC0000 --> Read access protection setting (1 unit = 1, format xx0000)
(not already tested, but it may be) 0 : Password deactivated, 1 : UT04-A0 and UT04-A9 password protected, 2 : Whole UT04 menu is password protected.

# RD2000 --> Read speed code of exaust vent (1 unit = 1, format xxxx00)
                   000000 --> Idle speed
                   000100 --> "Power 1" speed
                   000200 --> "Power 2" speed
                   000300 --> "Power 3" speed
                   000400 --> "Power 4" speed
                   000500 --> "Power 5" speed
                   000600 --> "Load wood" speed
                   000700 --> "Fire on" speed
                   000800 --> "Stop Fire" (burn pot cleaning) speed / "Stop" cooling procedure speed

# RD3000 --> Read speed code of ambient vent (1 unit = 1, format xxxx00)
                   000100 --> "Power 1" speed
                   000200 --> "Power 2" speed
                   000300 --> "Power 3" speed
                   000400 --> "Power 4" speed
                   000500 --> "Power 5" speed

# RD4000 --> Read speed code of pellet snail (1 unit = 1, format xxxx00)
                   000000 --> Idle speed
                   000100 --> "Power 1" speed / "Stop Fire" (burn pot cleaning) speed
                   000200 --> "Power 2" speed
                   000300 --> "Power 3" speed
                   000400 --> "Power 4" speed
                   000500 --> "Power 5" speed
                   000600 --> "Load wood" speed
                   000700 --> "Fire on" speed

# RD7000 --> Read water temperature (1 unit = 1°C, format xxxx00)

# RD8000 --> Read countdown cycle : time left until next burn pot cleaning or cleaning time left when cleaning burn pot (1 unit = 1min or 1sec, format mm:ss00)

# RD9000 --> Read sequence cycle and status code (1 unit = 1, format xxyy00)
Stop sequence terminated :
  000000 --> Seq 00 stat 00 --> Phase 0 Stop; display "off"
Startup sequence :
  010100 --> Seq 01 stat 01 --> Phase 1 Igniter/resistor ON/Preheat, exhaust fan test; display "fan acc"
  010200 --> Seq 01 stat 02 --> Phase 2 Igniter/resistor ON, pellet motor cycling; display "load wood"
  010300 --> Seq 01 stat 03 --> Phase 3 Flame stabilisation; display "fire on"
  010400 --> Seq 01 stat 04 --> Phase 4 Flame stabilisation, Igniter/resistor OFF; display "fire on"
Running sequence :
  020100 --> Seq 02 stat 01 --> Phase 1 Startup finished, heating; display sequentially actual temp, mode (auto), actual power
Cleaning sequence :
  040100 --> Seq 04 stat 01 --> Phase 1 Cleaning burn pot; display "fire off" 
Cooling sequence (normal mode) :
  080100 --> Seq 08 stat 01 --> Cooling phase 1, restart not possible; display "atte"
  080200 --> Seq 08 stat 02 --> Cooling phase 2, restart possible (Fume actual temp below setting in menu UT04-A0-Pr37)
Stand-by sequence (UT04-A9-Pr28 with a value in 1 °C steps, e.g. if Pr28 = 5, stove stops at T° set + 5°C & restarts at T° set -5°C) :
  100100 --> Seq 10 stat 01 --> Cooling ph. 1, restart not possible; display "atte"
  100200 --> Seq 10 stat 02 --> Cooling ph. 2, restart possible (Fume actual temp below setting in menu UT04-A0-Pr37)
  100300 --> Seq 10 stat 03 --> Off  (stand-by mode terminated/waiting restart according Pr28 hysteresis)

# RDA000 --> Read Error code (1 unit = 1, format xxxx00)

                   000000 --> No error
                   000100 --> Ignition failure
                   000200 --> Defective suction sensor / Malfunction of Fume Extractor Engine
                   000300 --> Insufficient air intake (Only if aspiration sensor present)
                   000400 --> Water temperature overheat
                   000500 --> Pellets finished
                   000600 --> Thermal security switch/Pressure security switch
                   000700 --> Thermal security switch/Pressure security switch (same as error 6 !?)
                   000800 --> Electrical black-out
                   000900 --> Exhaust motor failure
                   000A00 --> Stove control board overtemperature / alarm code 10
                   000B00 --> Date expired / alarm code 11
                   000C00 --> alarm code 12
                   000D00 --> alarm code 13
                   000E00 --> Overheating / alarm code 14
                   000F00 --> -- no alarm message ---

# RDB000 --> Read sucking sensor level (1 unit = 1, format xxxx00, for analog sensor = level, for digital/switch sensor = 256 ok or 768 alarm)

# RDC000 --> Read firmware version [returns 9270 for 9.2.7] (1 unit = 1, format x.x.x000)

# RDD000 --> Read exhaust motor speed correction (1 unit = 10RPM, format xx0000)

# RDE000 --> Read pellet snail speed correction (1 unit = 1%, format xx0000) 

# RDF000 --> Read PCB temperature (1 unit = 1°C, format xxxx00)

# RED000 --> Read total hours of running (1 unit = 1h, format xxxxxx) == menu UT04-Pr02

# REE000 --> Read partial hours of running since last servicing (1 unit = 1h, format xxxxxx) == menu UT04-Pr01

# REF000 --> Read exhaust fan round per minute (1 unit = 10RPM, format xxxx00)

And finally a helpful technical french documentation, even it's not the same controller : 

https://fr.scribd.com/document/817671132/MANUALE-TERMOVANA-SCHEDA-VM88-3-TASTI-FR-MARIA2-0-1

Have a nice day.

aceindy, 2025-03-19:
Awesome...too bad I no longer own the stove 😢
But maybe @jbjardine Can do something useful with this.
It could also be interesting for @mvroosmalen1970 😁

mvroosmalen1970, 2025-03-19:
Hi,
 
Already implemented most of these a while ago ;-) you can see it working in ESPhome. Let me know if you have any issues.  
 
Mark

> @aceindy (https://github.com/aceindy)   @jbjardine (https://github.com/jbjardine)  For your information -- USE AT OWN RISK --- (Tested on controller BOARD V9 AMG 410615 V9.2.7 made by Duepi, Duepi [...]
               000100 -->  "Power 1" speed
               000200 -->  "Power 2" speed
               000300 -->  "Power 3" speed
               000400 -->  "Power 4" speed
               000500 -->  "Power 5" speed
               000600 -->  "Load wood" speed
               000700 -->  "Fire on" speed
               000800 -->  "Stop Fire" (burn pot cleaning) speed / "Stop" cooling procedure speed

> RD3000 -->  Read speed code of ambient vent (1 unit = 1, format xxxx00)
>                000100 -->  "Power 1" speed
               000200 -->  "Power 2" speed
               000300 -->  "Power 3" speed
               000400 -->  "Power 4" speed
               000500 -->  "Power 5" speed

> RD4000 -->  Read speed code of pellet snail (1 unit = 1, format xxxx00)
>                000000 -->  Idle speed
               000100 -->  "Power 1" speed / "Stop Fire" (burn pot cleaning) speed
               000200 -->  "Power 2" speed
               000300 -->  "Power 3" speed
               000400 -->  "Power 4" speed
               000500 -->  "Power 5" speed
               000600 -->  "Load wood" speed
               000700 -->  "Fire on" speed

> RD7000 -->  Read water temperature (1 unit = 1°C, format xxxx00)
> RD8000 -->  Read countdown cycle : time left until next burn pot cleaning or cleaning time left when cleaning burn pot (1 unit = 1min [...]
000000 -->  Seq 00 stat 00 -->  Phase 0 Stop; display "off"
Startup sequence :
010100 -->  Seq 01 stat 01 -->  Phase 1 Igniter/resistor ON/Preheat, exhaust fan test; display "fan acc"
010200 -->  Seq 01 stat 02 -->  Phase 2 Igniter/resistor ON, pellet motor cycling; display "load wood"
010300 -->  Seq 01 stat 03 -->  Phase 3 Flame stabilisation; display "fire on"
010400 -->  Seq 01 stat 04 -->  Phase 4 Flame stabilisation, Igniter/resistor OFF; display "fire on"
Running sequence :
020100 -->  Seq 02 stat 01 -->  Phase 1 Startup finished, heating; display sequentially actual temp, mode (auto), actual power
Cleaning sequence :
040100 -->  Seq 04 stat 01 -->  Phase 1 Cleaning burn pot; display "fire off"
Cooling sequence (normal mode) :
080100 -->  Seq 08 stat 01 -->  Cooling phase 1, restart not possible; display "atte"
080200 -->  Seq 08 stat 02 -->  Cooling phase 2, restart possible (Fume actual temp below setting in menu UT04-A0-Pr37)
Stand-by sequence (UT04-A9-Pr28 with a value in 1 °C steps, e.g. if Pr28 = 5, stove stops at T° set + 5°C  &  restarts at T° set -5°C) :
100100 -->  Seq 10 stat 01 -->  Cooling ph. 1, restart not possible; display "atte"
100200 -->  Seq 10 stat 02 -->  Cooling ph. 2, restart possible (Fume actual temp below setting in menu UT04-A0-Pr37)
100300 -->  Seq 10 stat 03 -->  Off (stand-by mode terminated/waiting restart according Pr28 hysteresis)
>
> RDA000 -->  Read Error code (1 unit = 1, format xxxx00)
>                000000 -->  No error
               000100 -->  Ignition failure
               000200 -->  Defective suction sensor / Malfunction of Fume Extractor Engine
               000300 -->  Insufficient air intake (Only if aspiration sensor present)
               000400 -->  Water temperature overheat
               000500 -->  Pellets finished
               000600 -->  Thermal security switch/Pressure security switch
               000700 -->  Thermal security switch/Pressure security switch (same as error 6 !?)
               000800 -->  Electrical black-out
               000900 -->  Exhaust motor failure
               000A00 -->  Stove control board overtemperature / alarm code 10
               000B00 -->  Date expired / alarm code 11
               000C00 -->  alarm code 12
               000D00 -->  alarm code 13
               000E00 -->  Overheating / alarm code 14
               000F00 -->  -- no alarm message ---
[... 66 lines cut ...]
Cooling sequence (normal mode) :
080100 -->  Seq 08 stat 01 -->  Cooling phase 1, restart not possible; display "atte"
080200 -->  Seq 08 stat 02 -->  Cooling phase 2, restart possible (Fume actual temp below setting in menu UT04-A0-Pr37)
Stand-by sequence (UT04-A9-Pr28 with a value in 1 °C steps, e.g. if Pr28 = 5, stove stops at T° set + 5°C  &  restarts at T° set -5°C) :
100100 -->  Seq 10 stat 01 -->  Cooling ph. 1, restart not possible; display "atte"
100200 -->  Seq 10 stat 02 -->  Cooling ph. 2, restart possible (Fume actual temp below setting in menu UT04-A0-Pr37)
100300 -->  Seq 10 stat 03 -->  Off (stand-by mode terminated/waiting restart according Pr28 hysteresis)
>
> RDA000 -->  Read Error code (1 unit = 1, format xxxx00)
>                000000 -->  No error
               000100 -->  Ignition failure
               000200 -->  Defective suction sensor / Malfunction of Fume Extractor Engine
               000300 -->  Insufficient air intake (Only if aspiration sensor present)
               000400 -->  Water temperature overheat
               000500 -->  Pellets finished
               000600 -->  Thermal security switch/Pressure security switch
               000700 -->  Thermal security switch/Pressure security switch (same as error 6 !?)
               000800 -->  Electrical black-out
               000900 -->  Exhaust motor failure
               000A00 -->  Stove control board overtemperature / alarm code 10
               000B00 -->  Date expired / alarm code 11
               000C00 -->  alarm code 12
               000D00 -->  alarm code 13
               000E00 -->  Overheating / alarm code 14
               000F00 -->  -- no alarm message ---

closed 2025-04-02 by aceindy