aceindy/Duepi_EVO #62: Current température si 0

Yosh76, 2025-02-14:
Hello ! Thanks a lot for your job, it’s a very good job ! 

I use Wemos d1 flash with ESP Easy Mega 
I don’t Forget 5 ms to RX. 

With home assistant or mydpremote, current temperature is always 0. 
Target temperature is ok. 

I don’t know what happens…
Wemos d1 is thé problem ?

Thanks.

aceindy, 2025-02-15:
Hi Yosh,

Not sure what the issue could be.

Are we talking about the current setpoint, or the current ambient temperature?
I know for sure not all stove support the current setpoint (mine doesn't).

That is the reason why it has the config entry:     temp_nofeedback: 16
So it will set the temp to, so just to be safe....

You can turn on the debug logging, and see what it says:

2025-02-15 08:36:08.392 DEBUG (MainThread) [custom_components.duepi_evo.climate] Setpoint retrieval not supported by this stove, using temp_nofeedback 16
2025-02-15 08:37:09.912 DEBUG (MainThread) [custom_components.duepi_evo.climate] Pellet stove: Received burner: Ignition starting, Ambient temp: 21.4, Fan speed: 1, Flu gas temp: 59, Exh fan speed: 2820, PelletSpeed: 0, Error code: All OK, **Setpoint temp: None**

Yosh76, 2025-02-16:
I speak about current ambient temperature. 
Always 0 to mydpremote and home assistant. 

Impossible to use ESPLink, I can’t flash second firmware user2.bin…

How can I turn on the debug logging please ?

aceindy, 2025-02-16:
I am actually wondering if that user2.bin is required...I can't remember 🤔 

And for logging....
https://github.com/aceindy/Duepi_EVO?tab=readme-ov-file#troubleshooting

aceindy, 2025-02-16:
@Yosh76 I just checked, the ESPLink should work fine 'out-of'the-box'
So no need to use user2.bin; (if I remember correctly, that is when upgrading an existing ESPLink, so not starting from scratch, not sure why someone would do that though........not much to save on an ESPLink 😉  )

Yosh76, 2025-02-16:
So, i get this code :

`2025-02-16 22:43:24.486 DEBUG (MainThread) [custom_components.duepi_evo.climate] granule: Received burner: Off, Ambient temp: 0.0, Fan speed: 0, Flu gas temp: 15, Exh fan speed: 0, PelletSpeed: 0, Error code: All OK, Setpoint temp: 19
`

aceindy, 2025-02-17:
To me it looks simply the stove reports 0.0

Since the interface is nothing more then a serial2lan converter, I suspect the stove doesn't send (the proper) value.
Since all other values seem to be OK, I doubt the issue is in the converter,

You could overcome this by using a 3rd party temperature sensor ?

Yosh76, 2025-02-17:
Ok. 
I already use an other temperature sensor. 
With Home assistant, it’s simple. 

But i’m like when all it’s ok ! 👍🏼 
Do you think it’s possible it’s Esp easy mega the problem ?
I will test with ESPLink with my Wemos. 
Nothing to do ? Only change gpi for RX / TX ?

Thanks !!

aceindy, 2025-02-17:
Yeah, only to configure RX/TX pins...here are mine (using ESP-01):

[image]

aceindy, 2025-02-27:
Closing this one, as it is probably not an issue of the integration, as MyDPremote has the same issue.
This appears to be an issue on the stove.

closed 2025-02-27 by aceindy